RenatoLAG/Braco-robotico #2: Procurar eventos para publicação de artigos

RenatoLAG, 2018-01-05:
Pesquisar e encontrar possíveis locais para publicação  de artigos científicos (feiras, congressos...)

hyperiondeimos, 2018-01-09:
não vi nada até agora.

RenatoLAG, 2018-01-26:
Eu, como professor, gostaria que todos os projetos tivessem significância para gerar boas publicações

O projeto proposto vem trazendo a utilização de técnicas de grande relevância, principalmente para área de pesquisa, com isso acredita-se que este projeto será de grande apreciação pela classe acadêmica, despertando o interesse de outras pessoas que buscam aprender mais sobre as tecnologias envolvidas neste projeto. Enfim, este projeto também pode servir de base para futuras melhorias.