RenatoLAG/Braco-robotico #12: Artigos correlatos

RenatoLAG, 2018-01-13:
Pesquisar artigos que sirvam como base dando ideias para a elaboração do projeto e escrita do artigo final

Suzielen, 2018-01-13:
Artigos usados para ajudar na elaboração do artigo e desenvolvimento do projeto: 
[2_7Engevista04.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628788/2_7Engevista04.pdf)
[214130.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628789/214130.pdf)
[ApresentacaoRaspberryPi.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628790/ApresentacaoRaspberryPi.pdf)
[Comportamentos_reativos_para_seguir_pistas_em_um_r.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628791/Comportamentos_reativos_para_seguir_pistas_em_um_r.pdf)
[ENEGEP2001_TR10_0917.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628792/ENEGEP2001_TR10_0917.pdf)
[LM_100_60_65_06_tut_raspberry_pi.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628793/LM_100_60_65_06_tut_raspberry_pi.pdf)
[PB_COENC_2014_1_02.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628794/PB_COENC_2014_1_02.pdf)
[Robô modular utilizado como limpador de praia.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628795/Robo.modular.utilizado.como.limpador.de.praia.pdf)
[Rubens Henrique Pailo de Almeida.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628796/Rubens.Henrique.Pailo.de.Almeida.pdf)
[visao computacional.pdf](https://github.com/RenatoLAG/Braco-robotico/files/1628797/visao.computacional.pdf)

hyperiondeimos, 2018-01-23:
Faltou adicionar a revisão sistemática. Somente colocar os artigos não é a revisão. Aonde estão os comentários e o entendimento de cada artigo?

RenatoLAG, 2018-01-23:
@hyperiondeimos professor, colocamos em outro issue chamado Revisão Sistemática

closed 2018-02-05 by hyperiondeimos