RenatoLAG/Braco-robotico #13: Revisão sistemática

RenatoLAG, 2018-01-13:
Após os artigos serem encontrados, realizar a revisão sistemática afim de verificar melhorias que possam ser feitas no nosso projeto e identificar quais métodos de pesquisa foram utilizados para auxiliar na escrita do relatório final.

TaynaM, 2018-01-14:
Aqui está a revisão sistemáticas dos artigos encontrados pela Suzielen e por mim.
[Eventos.docx](https://github.com/RenatoLAG/Braco-robotico/files/1628948/Eventos.docx)

TaynaM, 2018-01-14:
Abaixo encontra-se a revisão sistemática dos artigos encontrados pela Suzielen e por mim.
[Fichamentos finalizados.docx](https://github.com/RenatoLAG/Braco-robotico/files/1628950/Fichamentos.finalizados.docx)

hyperiondeimos, 2018-01-23:
Tem algumas falhas, mas tudo bem. Darei como fechado.

hyperiondeimos, 2018-01-23:
Só enviem da próxima vez em PDF.

closed 2018-01-23 by hyperiondeimos