RenatoLAG/Braco-robotico #16: Vídeos de simulação

RenatoLAG, 2018-01-14:
Postar os vídeos que serviram de base para a escolha de certas características do projeto

TaynaM, 2018-01-14:
Aqui estão alguns dos vídeos pesquisados para ajudar na execução do projeto
[https://www.youtube.com/watch?v=heGqTK1aqK4](url)
[https://www.youtube.com/watch?v=XkqY4zTwsm4](url)
[https://www.youtube.com/watch?v=7UsmJgHU6wk](url)

Suzielen, 2018-01-15:
Alguns links para ajudar no projeto...
Compilação e instalação da Raspberry Pi:
https://www.youtube.com/watch?v=RE-p7GedP2Y
Visão Computacional com Python e OpenCV:
https://www.youtube.com/watch?v=lmJxGmOAtq8
Introdução à Biblioteca de Visão Computacional:
https://www.youtube.com/watch?v=-Hb7ZFSYqnk

hyperiondeimos, 2018-01-23:
inserir na sprint #2

hyperiondeimos, 2018-01-23:
Irei fechar isso depois da reunião, ou mudem os dados dos releases.

TaynaM, 2018-02-04:
Como não fizemos mais o uso da visão computacional e nem da Raspberry no projeto final, os vídeos ficaram sem utilidade. Portanto, a tarefa será congelada.