RenatoLAG/Braco-robotico #18: Aviso geral

TaynaM, 2018-01-15:
Gente, é o seguinte, todas as atividades propostas aqui são essenciais para o desenvolvimento do projeto e devem ser realizadas. Então fiquem sempre atentos e tentem comentar para trocarmos ideias.

hyperiondeimos, 2018-01-23:
@TaynaM Nenhuma pessoa da sua equipe está participando do Scrum. Você como ScrumMaster deve fazer o Scrum funcionar. Hoje gostaria de ter uma reunião com vocês, a tarde.

hyperiondeimos, 2018-01-23:
Ainda falta e muito quebrar os épicos em mais tarefas....

RenatoLAG, 2018-01-23:
Estaremos lá, professor

vrlimaa, 2018-02-02:
[image]

Aviso geral! Alteração no projeto final idealizado.

Observações gerais:
1. Foi descartado a utilização da câmera como visão computacional.
2. Agora utilizaremos o artifício do sensor óptico.
3. Implementaremos o CLP no atuador. 
4. De formar geral o projeto terá os mesmos moldes de eficiência para antiga idealização.
5. Croqui oriunda da ideia proposta pelo próprio professor.

“Projeto do braço robótico: quero que pelo menos o projeto consiga contar a quantidade de peças que passam e que ele possa separar as peças por contagem, tipo, a primeira peça detectada passa direto na esteira, a segunda o braço pega e guarda e a terceira o pistão pneumático empurra e somente assim o ciclo se repete.”