RenatoLAG/Braco-robotico #20: Confecção da esteira

TaynaM, 2018-01-15:
Disponibilizar aqui o modelo, o material e as dimensões da esteira.

hyperiondeimos, 2018-01-23:
Fotos e montagem, aonde estão? Além disso, vocês devem criar um WIKI com todos os passos da montagem.

vrlimaa, 2018-01-23:
Tayna, como havíamos combinado na última reunião presencial, ontem na segunda 23, consegui fazer a compra no mercado dos últimos produtos que faltavam para a montagem da esteira.

vrlimaa, 2018-01-23:
[image]

vrlimaa, 2018-01-23:
Materiais utilizados

Cano PVC de 25 mm
Varios pedaços de madeira
Lona de borracha 
Cola quente 
Palito de picolé
Cola mil
Parafusos de diversos tamanhos 
Rosca pra parafuso 
Motor dc 12v (utilizado usualmente em impressora) 
Engrenagem de plástico
Cerra copo
Cerrote
Furadeira
Parafusadeira

vrlimaa, 2018-01-23:
A montagem da mesma se iniciará hoje!

vrlimaa, 2018-01-23:
[image]

vrlimaa, 2018-01-23:
A ideia da esteira é essa transportadora que havíamos decidido em comum acordo, e ao final será repassada a planilha com o dimensionado das demais peças, para um entendimento e visualização mais simplista, como acordado.

vrlimaa, 2018-01-23:
E conforme for o processo de montagem, irei postando a foto das montagens das etapas já concluídas.

vrlimaa, 2018-01-23:
Peço desculpas ao professor, pois não me atentei a adesão da minha assinatura ao grupo, e fiquei por hora ficando um pouco de "escanteio" da sprint e não realizando o processo de desenvolvimento desejado da disciplina.

vrlimaa, 2018-01-23:
Porém o processo não encontra-se estagnado como pode-se observar, encontra-se em fase de execução da montagem nesse momento, conforme for irá sendo postado o desenvolvimento do mesmo, como já relatado.

vrlimaa, 2018-01-23:
Lembrando que vale ressaltar que optamos por uma esteira plana e NÃO inclinada, como se encontrada no modelo de referência, de modo geral a esteira estará nos mesmos moldes dessa que ser como base de referência, porém ela será na disposição plana e não inclinada como está na foto.

TaynaM, 2018-01-24:
Tudo bem, Vinícius.  Peço que tu faça o orçamento e disponibilize na issue de Orçamento e custo para termos uma noção do custo total do projeto e dividirmos igualmente os valores após o término do projeto.

vrlimaa, 2018-01-25:
Boa noite a todos, hoje pela tarde estive em contato com o professor, onde o mesmo muito solicito nos disponibilizou novos materiais e até mesmo ideias, para a confecção da esteira, portanto, houve alteração, claro que no sentido de melhoramento para o protótipo final da esteira, desta vez ela será produzida em uma base toda de alumínio, com rolamentos profissionais para um melhor atrito e um motor relativamente melhor em todos os sentidos, principalmente na força. Sendo assim, com as melhorias não encontraremos problemas que estavam atrasando o andamento do restante do projeto, pois a esteira além de ficar mais apresentável, irá também nos trazer uma serventia de 100% do idealizado, irei postar as fotos dos materiais disponibilizados por ele.

vrlimaa, 2018-01-25:
[image]
 Aqui é a lona, disponibilizada pelo professor, vale ressaltar que ela além de ser muito resistente, irá compreender uma aderência muito mais aceitável na esteira.

vrlimaa, 2018-01-25:
[image]

Esse é o motor, motor encontrado em vidro elétrico de automóvel, sendo assim tendo um torque muito mais interessante para o nosso interesse, assim ele irá suprir toda nossa necessidade para a questão da esteira. OBS: Motor 12 Volts.

vrlimaa, 2018-01-25:
[image]

A base de alumínio, parafusos, porcas e rolamentos (os que irão ser aderidos pela lona disponibilizado por ele) e esses nas caixas serão os internos que fiz a compra hoje a tarde mesmo, em uma loja de indicação do professor, a ROLAPEÇAS, loja de rolamento e equipamentos diversos para mecânica que se encontra no centro, mais precisamente na AVENIDA TEFÉ.

vrlimaa, 2018-01-25:
NOTA 1: Fiz a compra de 4 rolamentos de 22mm de diametro que serão acoplados aos rolamento maiores, para que sejam introduzidos no motor e nos parafusos que darão a base da esteira.
NOTA 2: Chave combinada de 1/2 para enroscar/desenroscar as porcas.

OBS: As ferramentas restantes já detenho.

vrlimaa, 2018-01-25:
A esteira será parafusada em uma estrutura de madeira, ideia que peguei de um modelo que o professor nos disponibilizou lá para visualização, ou seja amanhã irei comprar a placa de madeira compeçada para preparar essa essa estrutura e sendo assim iniciar o processo de montagem da mesma, por hora, está tudo dentro do planejado, houve o contra tempo por conta da alteração do modelo, porém acredito que agora já está tudo resolvido e dentro do organizado para finalização dessa parte.

vrlimaa, 2018-01-25:
Por fim, agradeço ao professor @hyperiondeimos pela força, ajuda que ele nos deu hoje, fazendo total diferença na melhoria do projeto.

vrlimaa, 2018-01-25:
@TaynaM @Suzielen Está tudo ok e entendido? Pergunto por conta da questão do artigo!

vrlimaa, 2018-01-26:
Boa noite a todos do grupo, hoje conseguimos concluir a confecção da esteira conforme o idealizado, fui em um marceneiro e ele montou uma base para que eu pudesse montar a esteira nela, duas peças 20 x 70 cm, irei agora replicar os procedimentos de montagem e o material completo.

vrlimaa, 2018-01-26:
[image]

vrlimaa, 2018-01-26:
[image]

vrlimaa, 2018-01-26:
[image]

vrlimaa, 2018-01-26:
[image]

vrlimaa, 2018-01-26:
Bom, como se pode observar, temos as duas peças 20x70 centímetro que serviram de base, a peça vertical foi a sustenção do motor:

- Motor parafusado na parte traseira.
- Há um buraco vazado na peça vertical, para que o eixo do motor possa varar do outro lado da peça(parte frontal) e assim fazer o posicionamento da esteira ao seu lado.
- Rolamentos centralizados na mesma horizontalidade do eixo do motor.
- A roda de borracha acoplada no eixo do motor serve de aderência para que o processo de giro da esteira funcione de forma correta.
- Rolamentos acoplados à dois parafusos, um de cada lado(início e fim).
- Rolamento próximo ao motor encontra-se encostado à roda de borracha fazendo com que aja o contato com a lona da esteira e assim tendo a aderência e atrito entre lona e roda, fazendo a esteira girar.

vrlimaa, 2018-01-26:
OBS:
- Temos também uma chapa de alumínio que serve de auxílio como sustentação para a lona da esteira.
- Para testes foi utilizado uma fonte 12 volts.
- O motor como já dito é de vidro elétrico de carro e tem polaridade, caso inverta a ligação do seu circuito, o sentido do giro do seu eixo também irá mudar de direção.
- A esteira já encontra-se na posse do Paulo.

vrlimaa, 2018-01-26:
Materiais utilizados:

- Motor 12v de vidro elétrico de carro.
- 4 rolamentos de 22 mm de diâmetro.
- 2 conectores de aço(onde foi introduzido os rolamentos)
- Uma lona.
- Uma chapa de alumínio.
- 2 peças de madeira. 
- Uma rodinha de borracha. 
- 2 parafusos de 10 cm (para encaixar os rolamentos)
- 6 porcas.
- 2 peças de madeira de 30 mm (pois o motor é um pouco pesado, a peça sendo grossa, tem uma melhor sustentação).
- Parafusos de diversas polegadas e tamanhos.

vrlimaa, 2018-01-26:
@Luis1404 qualquer coisa me contactar.

vrlimaa, 2018-01-26:
Gostaria de fazer upload de um vídeo do funcionamento da esteira, porém não achei onde tem essa opção, só achei a opção fazer upload para foto, sendo assim irei repassar a todos via WhatsApp, para que assim consigam ver o funcionamento prático da esteira.

TaynaM, 2018-02-02:
A esteira que será utilizada não será mais essa devido as mudanças físicas na estrutura do projeto. A nova esteira foi disponibilizada pela instituição.

vrlimaa, 2018-02-02:
Por toda via, todas as etapas de montagem serão disponibilizadas, projeção mecânica, elétrica, desenhos procedemos, posteriormente, como se a mesma tivesse sido confeccionada por nós.

vrlimaa, 2018-02-04:
[image]

vrlimaa, 2018-02-04:
DESENHOS MECÂNICOS:

[image]
[image]
[image]

vrlimaa, 2018-02-04:
Materiais Utilizados no protótipo e quantidade: 

 BASE - 01
 MANCAL - 02
 MANCAL MOTOR CORREIA - 01
 MANCAL MOTOR - 01
 Rolling bearing GB/T 5800-2003-626 - 04
 ROLETE - 01
 ROLETE MOTOR - 01
 ESTEIRA BORRACHA - 01
 ROLDANA - 02

closed 2018-02-05 by hyperiondeimos