RenatoLAG/Braco-robotico #22: Programação do Arduino e da Raspberry

TaynaM, 2018-01-15:
Inserir aqui as programações do Arduino que irá controlar o braço e a parte de visão computacional que fará o reconhecimento dos objetos. Além disso, disponibilizar o material de apoio que está sendo utilizado.

hyperiondeimos, 2018-01-23:
Os dados de códigos não são inseridos aqui, ok. Deverão ser inseridos na aba <CODE>. Podem deixar aqui essa issue, mas deve passar apenas os commits

RenatoLAG, 2018-01-24:
Raspbian Stretch: Install OpenCV 3 + Python on your Raspberry Pi

https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

RenatoLAG, 2018-01-24:
Raspbian Stretch: Install OpenCV 3 + Python on your Raspberry Pi

https://www.youtube.com/watch?v=j6RD3X94rEA&feature=youtu.be

RenatoLAG, 2018-01-24:
RASPBERRY PI - INSTALANDO O RASPBIAN | Vídeo Aula #51

https://www.youtube.com/user/canalwrkits/search?query=RASPBERRY

RenatoLAG, 2018-01-24:
Braço Robótico com Arduino

https://www.youtube.com/watch?v=vVOyWQZ25M8&t=1270s

RenatoLAG, 2018-01-24:
SKETCH - 4 SERVOS:

closed 2018-01-24 by RenatoLAG