RenatoLAG/Braco-robotico #24: Reunião do artigo

TaynaM, 2018-01-24:
Meninas, temos que começar a escrita do artigo já que o mesmo é para semana que vem. Então, vamos nos reunir todos dias até a data da entrega do projeto. A primeira reunião será amanhã às 13:00.

vrlimaa, 2018-01-25:
Postei as atualizações referente a esteira e que pode servir de serventia, claro, para o artigo, qualquer dúvida, estou a disposição.

Rayara, 2018-01-25:
Muito obrigada Vinícius. Toda e qualquer descrição do projeto será de grande ajuda para a produção do artigo.

RenatoLAG, 2018-01-26:
enviarei ainda hoje (se possivel) o procedimento para a preparação da Raspberry pi3B na utilização para este projeto.

Rayara, 2018-02-01:
A colaboração para o desenvolvimento do artigo, está sendo realizada de forma presencial ou online.