RenatoLAG/Braco-robotico #25: Visão computacional

TaynaM, 2018-01-24:
Estudar o modo de funcionamento da visão computacional e sua aplicação no projeto. Após isso, disponibilizar os materiais de apoio.

RenatoLAG, 2018-01-24:
**Raspbian Stretch: Install OpenCV 3 + Python on your Raspberry Pi**

https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

https://www.youtube.com/watch?v=j6RD3X94rEA&feature=youtu.be

RenatoLAG, 2018-01-25:
instalação do **openCV** está temporariamente parada, devido questões pessoais. retomarei a terceira tentativa de instalação no sábado ou domingo.

RenatoLAG, 2018-01-25:
na primeira tentativa de instalação do **openCV** na **Raspberry pi 3B** houve um travamento em 83% no processo de instalação. em seguida o professor sugeriu que a instalação fosse feita remotamente através do notebook por meio da conexão com a internet tanto no notebook quanto na Raspberry, utilizando o programa **Putty**, essa foo a segunda tentativa, onde ocorreu um erro na conexão com a internet em 86% do processo de instalação.

hyperiondeimos, 2018-01-25:
Senhores, segue um Script feito para automatizar o processo de instalação do OpenCV no Raspberry Pi 2 e 3. Sigam isso com cuidado e prestem atenção. Leiam os artigos e tutoriais passados com clareza que vai dar tudo certo.

[image]

Os scripts estão na pasta de aula 07 no Dropbox!

RenatoLAG, 2018-01-25:
Vamos testar professor.

TaynaM, 2018-02-02:
Como não vamos mais utilizar a visão computacional no projeto, a tarefa será congelada.