RenatoLAG/Braco-robotico #26: Programação da Raspberry

TaynaM, 2018-01-24:
Comunicar todo o processo da implementação da Raspberry no projeto incluindo os drivers instalados, a comunicação com o arduino e tudo o mais.

RenatoLAG, 2018-01-24:
**RASPBERRY PI - INSTALANDO O RASPBIAN | Vídeo Aula #51**

https://www.youtube.com/watch?v=mEx4XEuYgYs&t=725s

**RASPBERRY PI - USANDO AS GPIOs | Vídeo Aula #61**

https://www.youtube.com/watch?v=GKK1B8gKU_E

RenatoLAG, 2018-01-24:
**SISTEMA OPERACIONAL**

DEBIAN STRETCH WITH RASPBERRY PI DESKTOP

http://downloads.raspberrypi.org/rpd_x86/images/rpd_x86-2017-12-01/2017-11-16-rpd-x86-stretch.iso

RenatoLAG, 2018-01-24:
**Raspberry Pi - How to SSH - Remote view your Pi...**

https://www.youtube.com/watch?v=ZkvHHkWmKSY

RenatoLAG, 2018-01-24:
**Download PuTTY**
PuTTY is an SSH and telnet client, developed originally by Simon Tatham for the Windows platform. PuTTY is open source software that is available with source code and is developed and supported by a group of volunteers.

https://www.putty.org/

TaynaM, 2018-02-02:
Como não vamos mais utilizar a Raspberry no projeto, a tarefa será congelada.