RenatoLAG/Braco-robotico #36: Estrutura física do projeto

TaynaM, 2018-02-02:
Definir a estrutura física do projeto com todos os seus detalhes.

Suzielen, 2018-02-04:
O projeto ira´ dispor de uma esteira com seu funcionamento por um motor DC de 12v, um atuador pneumático , mangueiras pu 6mm, válvulas, sensor óptico pnp, CPL, Arduíno para controle do braço robótico, braço robótico em mdf, servo motores, chaves do tipo push botton para acionamento do sistema e protoboard.

Suzielen, 2018-02-04:
Disponibilizarei aqui imagens da estrutura física do projeto bem como alguns dos materiais usados: 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2018-02-05 by hyperiondeimos