RenatoLAG/Braco-robotico #43: Vídeo final

TaynaM, 2018-02-04:
Postar o vídeo de simulação realizado no dia da entrega do projeto.

vrlimaa, 2018-02-04:
[image]

Braço Robótico utilizado no projeto!

vrlimaa, 2018-02-04:
[image]

vrlimaa, 2018-02-04:
[image]

[image]

[image]

Teste de funcionamento do protótipo.

vrlimaa, 2018-02-04:
[image]

Teste de funcionamento do braço robótico.

vrlimaa, 2018-02-04:
[Vídeo do protótipo em funcionamento ](https://www.dropbox.com/s/tle6bzaik0xq80i/V%C3%ADdeo%201.mp4?dl=0)

vrlimaa, 2018-02-04:
AQUI ESTÁ TODOS OS COMPONENTES ESSENCIAIS PARA A CONFECÇÃO DO PROJETO!

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

hyperiondeimos, 2018-02-05:
Bom terem utilizado um CLP, isso deu uma nota a mais para vocês.

closed 2018-02-05 by hyperiondeimos